davidk/PrusaSlicer-ARM.AppImage #4: Segmentation fault

Bupher, 2020-10-06:
I get Segmentation faults when I try and run the AppImage on either of my Pi4s. One is a 1GB and one is a 2GB, running octopi and buster lite respectively. Do they need to be 4GB models to install, I didn't see that as a requirement anywhere on your page?

pi@bupherspi:~/download $ ./PrusaSlicer-2.2.0-armhf.AppImage
Segmentation fault
pi@bupherspi:~/download $

davidk, 2020-10-07:
Hello there!

Its unfortunate that the error message didn't provide more of an explanation.

I believe that PrusaSlicer crashes attempting to find a GUI; it will need to be run with either its command-line options for slicing, or with the environmental variable `DISPLAY` set by a GUI.

Running without options defaults to launching the GUI (without `DISPLAY` set will result in a segmentation fault):

```
pi@pi4:~/Desktop $ echo $DISPLAY

pi@pi4:~/Desktop $ ./PrusaSlicer-2.2.0-armhf.AppImage 
Segmentation fault
```

CLI slicing should work still, if this is your desire:

```
$ ./PrusaSlicer-2.2.0-armhf.AppImage -s ~/Downloads/noctua-rpi4-poe.stl
10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill
88 => Generating skirt
90 => Exporting G-code to /home/pi/Downloads/noctua-rpi4-poe.gcode
Slicing result exported to /home/pi/Downloads/noctua-rpi4-poe.gcode
```

Aside from very large models that might make PrusaSlicer consume large amounts of memory, I don't forsee an issue with the AppImage being run on Pi 4s smaller than 4GB, but please let me know if is a limitation.

Thanks for filing this issue!

Bupher, 2020-10-07:
I opened up a VNC and it loaded it right up. I thought that was an installer, not the actual program. The end goal was to use it CLI, but It loaded right up.

Bupher, 2020-10-07:
And I just tried it through the CLI as well, with the desktop disabled, and it works as intended. Thanks, so much!

closed 2020-10-07 by Bupher