davidk/PrusaSlicer-ARM.AppImage #6: 64-bit Buster

koendv, 2020-11-03:
Thanks for providing this AppImage.
```
 $ uname -a
Linux raspberrypi 5.4.51-v8+ #1327 SMP PREEMPT Thu Jul 23 11:11:34 BST 2020 aarch64 GNU/Linux
 $ ./PrusaSlicer-2.3.0-alpha2-armhf.AppImage 
./PrusaSlicer-2.3.0-alpha2-armhf.AppImage: error while loading shared libraries: libz.so: cannot open shared object file: No such file or directory
```
Could you add a build for 64-bit raspberry os?

koendv, 2020-11-05:
Ended up compiling a backport of libcgal-dev_5.1.1 to buster/arm64. prusa-slicer 2.3.0 compiles and runs fine with that.

*Recipe:*
Following the procedure on [Debian Wiki](https://wiki.debian.org/SimpleBackportCreation):

```
sudo apt-get install packaging-dev debian-keyring devscripts equivs
```
Add "deb-src http://deb.debian.org/debian/ testing main" to /etc/apt/sources.conf
```
sudo apt update
sudo apt source libcgal-dev/testing
cd cgal-5.1/
sudo  mk-build-deps --install --remove
dch --bpo
fakeroot debian/rules binary
```
Remove "deb-src http://deb.debian.org/debian/ testing main" from /etc/apt/sources.conf

*Install backported libcgal-dev package*
```
cd ..
sudo dpkg -i ./libcgal-dev_5.1-1~bpo10+1_arm64.deb
```

koendv, 2020-11-06:
OK. Thank you for your 64-bit prusa-slicer AppImage.

I've done a clean install of 2020-08-20-raspios-buster-arm64, and downloaded the AppImage, and done a 
```
sudo apt-get install -y git cmake libboost-dev libboost-regex-dev libboost-filesystem-dev \
> libboost-thread-dev libboost-log-dev libboost-locale-dev libcurl4-openssl-dev libwxgtk3.0-dev \
> build-essential pkg-config libtbb-dev zlib1g-dev libcereal-dev libeigen3-dev libnlopt-cxx-dev \
> libudev-dev libopenvdb-dev libboost-iostreams-dev libnlopt-dev libdbus-1-dev
```
Starting the AppImage gives the following error message:
```
prusa-slicer: error while loading shared libraries: libgmpxx.so.4: cannot open shared object file: No such file or directory
```
so I do an additional 
```
sudo apt-get install libgmpxx4ldbl
```
Trying again:

```
$ ./PrusaSlicer-2.3.0-alpha3-arm64.AppImage 
PrusaSlicer GUI initialization failed: boost::filesystem::canonical: No such file or directory: "/home/pi/Downloads/prusa-slicer"

(process:2873): GLib-GObject-ESC[1;33mWARNINGESC[0m **: ESC[34m23:04:16.390ESC[0m: invalid (NULL) pointer instance

(process:2873): GLib-GObject-ESC[1;35mCRITICALESC[0m **: ESC[34m23:04:16.390ESC[0m: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:2873): Gtk-ESC[1;35mCRITICALESC[0m **: ESC[34m23:04:16.392ESC[0m: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:2873): GLib-GObject-ESC[1;35mCRITICALESC[0m **: ESC[34m23:04:16.393ESC[0m: g_object_get: assertion 'G_IS_OBJECT (object)' failed
```
I'm not certain what happens. I've done the following:
```
$ strace ./PrusaSlicer-2.3.0-alpha3-arm64.AppImage 2>&1 | tee /tmp/log
$ grep prusa-slicer /tmp/log
prusa-slicer: error while loading shared libraries: libnlopt.so.0: cannot open shared object file: No such file or directory
prusa-slicer: error while loading shared libraries: libboost_log.so.1.67.0: cannot open shared object file: No such file or directory
prusa-slicer: error while loading shared libraries: libgmpxx.so.4: cannot open shared object file: No such file or directory
PrusaSlicer GUI initialization failed: boost::filesystem::canonical: No such file or directory: "/home/pi/Downloads/prusa-slicer"
read(4, "prusa-slicer\n", 128)          = 13
execve("/tmp/.mount_PrusaS8laLiC/usr/bin//prusa-slicer", ["prusa-slicer"], 0x55aa788108 /* 56 vars */) = 0
faccessat(AT_FDCWD, "/tmp/.mount_PrusaS8laLiC/home/pi/Downloads/prusa-slicer", F_OK) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/pi/Downloads/prusa-slicer", 0x7fcc5ec1c8, 0) = -1 ENOENT (No such file or directory)
write(2, "boost::filesystem::canonical: No"..., 90boost::filesystem::canonical: No such file or directory: "/home/pi/Downloads/prusa-slicer") = 90
PrusaSlicer GUI initialization failed: boost::filesystem::canonical: No such file or directory: "/home/pi/Downloads/prusa-slicer"
read(4, "prusa-slicer\n", 128)          = 13
execve("/tmp/.mount_PrusaSoN30SG/usr/bin//prusa-slicer", ["prusa-slicer"], 0x5575bfec38 /* 33 vars */) = 0
faccessat(AT_FDCWD, "/tmp/.mount_PrusaSoN30SG/home/pi/Downloads/prusa-slicer", F_OK) = -1 ENOENT (No such file or directory)
newfstatat(AT_FDCWD, "/home/pi/Downloads/prusa-slicer", 0x7fe2214fc8, 0) = -1 ENOENT (No such file or directory)
write(2, "boost::filesystem::canonical: No"..., 90boost::filesystem::canonical: No such file or directory: "/home/pi/Downloads/prusa-slicer") = 90
```

So I'm wondering whether AppRun finds the correct prusa-slicer binary?

I've got the feeling we're on the right track; but we're not there yet. Again thanks for the effort so far.

davidk, 2020-11-06:
Hello!

Apologies for not leaving a note when the amd64 AppImage was generated, I believe I had just missed the update to this issue by a few minutes!

I also ran into your latter issue after building the AppImage. Can you try running `mkdir /home/pi/Downloads/prusa-slicer"`? Based on this line: `PrusaSlicer GUI initialization failed: boost::filesystem::canonical: No such file or directory: "/home/pi/Downloads/prusa-slicer"`

koendv, 2020-11-06:
Hi David.
Did a
`mkdir /home/pi/Downloads/prusa-slicer`.
Now runs. Slices. 
Good one.

closed 2020-11-15 by davidk