davidk/PrusaSlicer-ARM.AppImage #10: Dependency error on version 2.3.0armhf

courtpl, 2021-01-14:
I have deployed the last 2.3.0-armhf.Appimage
And executing the mommand:
sudo apt-get install -y git cmake libboost-dev libboost-regex-dev libboost-filesystem-dev \
libboost-thread-dev libboost-log-dev libboost-locale-dev libcurl4-openssl-dev libwxgtk3.0-dev \
build-essential pkg-config libtbb-dev zlib1g-dev libcereal-dev libeigen3-dev libnlopt-cxx-dev \
libudev-dev libopenvdb-dev libboost-iostreams-dev libnlopt-dev libdbus-1-dev imagemagick \
libgtk-3-dev libwxgtk3.0-gtk3-dev

But i have this error on app startup:
prusa-slicer: error while loading shared libraries: libgmpxx.so.4: cannot open shared object file: No such file or directory

The application doesn't not start.

davidk, 2021-01-14:
Hello,

Thank you for the detailed report, and apologies for the missing dependency. It looks like the package `libgmpxx4ldbl` should fulfill that dependency. Could you please run `sudo apt-get install libgmpxx4ldbl -y` and then start PrusaSlicer again to see if it resolves the issue?

courtpl, 2021-01-14:
I have tried installing this dependency, now i't work fine.
Thanks for your quick answwer.

Regards

commit 86ccd4f5 (2021-01-15): build/64build: Update dependencies

This adds libgtk2.0-dev (for linking through pkg-config)
and libgmpxx4ldbl for #10

resolves #10
closed 2021-01-15 by davidk