davidk/PrusaSlicer-ARM.AppImage #12: Fuse error?

McDabbs, 2021-01-20:
So I believe I followed the directions correctly (I am still relatively new with linux/raspberry pi's) but when attempting to open (using ./PrusaSlicer-2.3.0-GTK2-armhf.AppImage) I get this error 

`pi@raspberrypi:~/Downloads $ ./PrusaSlicer-2.3.0-GTK2-armhf.AppImage
fuse: device not found, try 'modprobe fuse' first

Cannot mount AppImage, please check your FUSE setup.
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
open dir error: No such file or directory
pi@raspberrypi:~/Downloads $ 
`

davidk, 2021-01-20:
Hello,

Thanks for filing this issue. I haven't run into a FUSE related error before, but it sounds like the module hasn't loaded for some reason or fuse isn't installed on your system..

Could you please try running `sudo apt-get install fuse -y`, `modprobe fuse`, and then re-run the AppImage to see if that resolves the issue?

davidk, 2021-01-31:
closing due to inactivity -- if you are still encountering any troubles here, no worries, feel free to reply/re-open and we can try to sort stuff out

closed 2021-01-31 by davidk