davidk/PrusaSlicer-ARM.AppImage #13: Dependency error: Package libwxgtk3.0-dev is not available,.

MichaelNewham, 2021-01-28:
Hi David,
I have Ubuntu Mate 20.10 groovy with aarch64 / arm 64 architecture on Raspberry Pi 400
Passing 'uname -a' outputs >
`Linux RPi400-UbuMateArch64-2010 5.8.0-1013-raspi #16-Ubuntu SMP PREEMPT Thu Jan 14 06:28:38 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux`
While 'Screenfetch' command outputs >
`OS: Ubuntu 20.10 groovy
Kernel: aarch64 Linux 5.8.0-1013-raspi
Shell: zsh 5.8
DE: MATE 1.24.1`
etc..
Have tried both the 'PrusaSlicer-2.3.0-GTK2-aarch64.AppImage' and the 'PrusaSlicer-2.3.0-GTK23-aarch64.AppImage'. They have extra permissions set but return the following error >
`prusa-slicer: error while loading shared libraries: libboost_log.so.1.67.0: cannot open shared object file: No such file or directory`
When i try to install the required dependencies OR try to build using the build.sh script i get the following error >
`Package libwxgtk3.0-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source`
I see from this Stack Exchange Q&A ([](https://askubuntu.com/questions/1241217/package-libwxgtk3-0-dev-has-no-installation-candidate-on-ubuntu-20-04)) that > libwxgtk3.0-dev was changed to libgtk3.0-gtk3-dev in 20.04. I have libgtk3.0-gtk3-dev already installed.
My source list at `/etc/apt/sources.list.d ` contains the following `deb http://ppa.launchpad.net/ubuntu-mozilla-security/ppa/ubuntu/ groovy main`.
Any idea how to work around this?
Cheers, Michael.

davidk, 2021-01-29:
Hello Michael,

It looks like Ubuntu Mate 20.10 has far later versions of all the packages that the AppImage requires to run, and it doesn't look like they're compatible at all. The earlier versions are not immediately available either without going through the trouble of manually installing each dependency, etc (but see below if you don't mind wrecking your system).

Adding to the issue, the build tooling is broken against 20.10 since `uname -i` reports a proper architecture (aarch64 on 20.10 versus unknown on Raspberry Pi 64-bit) so it can't be built against those versions.

I'm afraid there isn't a quick resolution to this issue other than attempting usage with one of the [Raspberry Pi OS 64-bit images](https://downloads.raspberrypi.org/raspios_arm64/images/), but i'll have to see if this is something that can be worked around in a more elegant manner in the future..

If you really do want to get this working on 20.10 (and would like to throw all caution to the wind), here is an approximate list of packages that would have to be downloaded and manually installed:

```
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-atomic1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-chrono1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-date-time1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-filesystem1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-regex1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-system1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-thread1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/i/icu/libicu63_63.1-6+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-log1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-locale1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/o/openvdb/libopenvdb5.2_5.2.0-5_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/b/boost1.67/libboost-iostreams1.67.0_1.67.0-13+deb10u1_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/i/ilmbase/libilmbase23_2.2.1-2_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/w/wxwidgets3.0/libwxgtk3.0-0v5_3.0.4+dfsg-8_arm64.deb
http://ftp.ie.debian.org/debian/pool/main/libj/libjpeg-turbo/libjpeg62-turbo_1.5.2-2+deb10u1_arm64.deb
```

MichaelNewham, 2021-01-29:
Tks for reply. I will backup the image before attempting your workaround. Will close the issue. Cheers!

closed 2021-01-29 by MichaelNewham

davidk, 2021-01-31:
Hi Michael,

I hope that this finds you well. With a minor set of workarounds PrusaSlicer compiles and is now able to run as an AppImage against 20.10. A built distribution can be found [here](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/download/version_2.3.0/PrusaSlicer-2.3.0-GTK3-arm64-Ubuntu-Mate-20.10.AppImage), if you would like to use it.

If for any reason you are interested in building your own AppImage to play around with things, the modified build utilities are located in the [20.10 branch](https://github.com/davidk/PrusaSlicer-ARM.AppImage/tree/20.10) of this repository -- the dependencies needed to run/build the AppImage can also be installed using the build script, but they are:

`git cmake libboost-dev libboost-regex-dev libboost-filesystem-dev libboost-thread-dev libboost-log-dev libboost-locale-dev libcurl4-openssl-dev build-essential pkg-config libtbb-dev zlib1g-dev libcereal-dev libeigen3-dev libnlopt-cxx-dev libudev-dev libopenvdb-dev libboost-iostreams-dev libgmpxx4ldbl libnlopt-dev libdbus-1-dev imagemagick libgtk2.0-dev libgtk-3-dev libwxgtk3.0-gtk3-dev`