davidk/PrusaSlicer-ARM.AppImage #14: 32 bit appimage segmentation fault

asperger514, 2021-02-19:
Hello, I'm a novice so forgive me if this turns out to be a waste of your time. I couldn't get the 32 bit appimage version to run without a segmentation fault without first copying the contents of my non-appimage installation located at  "/home/pi/Desktop/Prusa/resources/profiles" to the "/usr/local/share/PrusaSlicer/profiles". After all the files were copied over the 32bit appimage runs correctly.

[image]

sab0276, 2021-02-19:
I also got segmentation error when trying to install the AppImage

davidk, 2021-02-19:
@asperger514 Thanks for the report! I was able to reproduce this and it looks like its not picking up the profiles embedded within the AppImage. There is an assert that fails before that as well.. I'll have a closer look soon and will see if there is a solution that doesn't require a workaround.

```
../src/common/ctrlsub.cpp(232): assert "IsValid(n)" failed in GetClientData(): Invalid index passed to GetClientData(
(prusa-slicer:29360): Gtk-WARNING **: 09:48:06.425: gtk_window_set_titlebar() called on a realized window
[2021-02-19 09:49:06.471381] [0xb13bb010] [error]   stl_open_count_facets: Couldn't open /usr/local/share/PrusaSlicer/profiles/Creality/ender3_bed.stl for reading
Segmentation fault
```

@sab0276 Hello! Apologies for the issue. Some things to check that might help narrow down the cause of the issue: Is `$DISPLAY` populated? (check with `echo $DISPLAY` on the command-line) 

..and does running `./PrusaSlicer-2.3.0-GTK3-armhf.AppImage --help` produce help output (or does it return the same error straightaway)?

davidk, 2021-02-21:
Hello,
Every image for the latest release has been re-generated and re-uploaded. This should let PrusaSlicer find the printers that are embedded inside the AppImage.

The assert that occurs before the segfault is an unsolved issue; it looks like some data is being collected for it upstream.

sab0276, 2021-02-22:
Still getting the error when trying to install PrusaSlicer-2.3.0-GTK3-armhf.AppImage.  
[image]

sab0276, 2021-02-22:
> @sab0276 Hello! Apologies for the issue. Some things to check that might help narrow down the cause of the issue: Is `$DISPLAY` populated? (check with `echo $DISPLAY` on the command-line)
>
> ..and [...]

I just ran echo $DISPLAY did not return anything.
Running ./PrusaSlicer-2.3.0-GTK3-armhf.AppImage --help did return the help info right away.

davidk, 2021-02-22:
@sab0276 Thanks for the information! There was another user that had a similar experience in issue #4 some time ago..

It looks like your dependencies are ok, but PrusaSlicer looks for $DISPLAY during launch. When it is empty it will segfault -- $DISPLAY is typically is not filled over an SSH session.

There are two workarounds when using PrusaSlicer over SSH, using X11 forwarding to your local system (this is typically something like `ssh -X pi@raspberrypi.system.example.com /home/pi/PrusaSlicer-2.3.0-GTK3-armhf.AppImage`) or pre-pending $DISPLAY during the AppImage launch (this will put PrusaSlicer on the monitors attached to your Pi, but will not allow control over SSH): `DISPLAY=":0" ./PrusaSlicer-2.3.0-GTK3-armhf.AppImage`.

The other alternative use over SSH is to slice purely on the command-line with a GUI (as in issue #4).

sab0276, 2021-02-22:
Ah gotcha.  I was trying to install the program over SSH, but will be running it on the desktop.  I didn't realize running the AppImage ran the app, I just thought it installed the app to be run later.  So anytime I want to run the app I have to run ./PrusaSlicer-2.3.0-GTK3-armhf.AppImage from the directory I downloaded it to?

davidk, 2021-02-22:
There is a way to manually add one. Running these commands should place a launcher in the 'Graphics' category of the menu. Be sure to edit the `Exec=` line to fit your setup.

 	curl -SsL https://raw.githubusercontent.com/prusa3d/PrusaSlicer/master/resources/icons/	PrusaSlicer_128px.png | sudo dd of=/usr/share/pixmaps/PrusaSlicer_128px.png
	
	sudo tee /usr/share/applications/PrusaSlicer.desktop <<'EOF'
	[Desktop Entry]
	Name=PrusaSlicer
	Exec=/home/pi/Downloads/PrusaSlicer-2.3.0-GTK3-armhf.AppImage
	Icon=/usr/share/pixmaps/PrusaSlicer_128px.png
	Terminal=false
	Type=Application
	Categories=Graphics;3DGraphics;
	EOF

asperger514, 2021-02-22:
Thank you for all your work, I removed the superfluous workaround directory copy of settings from "/usr/local/share/PrusaSlicer/profiles" and the new appimage works great.

closed 2021-02-22 by asperger514

sab0276, 2021-02-22:
This worked great, thank you!