davidk/PrusaSlicer-ARM.AppImage #15: questions to building AppImages

bubnikv, 2021-03-15:
Hello David.

Thanks for all your effort building AppImages for ARMs. We at Prusa Research are now considering to integrate your effort into our build system, the main driver being adoption of chromebooks by US schools and the Linux on Chromebook effort, which is quite usable including OpenGL HW virtualization.

I suppose you are building your ARM AppImage builds on a R-PI.

Do you build both 32bit and 64 bit builds on the same R-Pi device running the same 64bit image? What image are you running?

Did you consider cross-compiling on an X86-64 box?

How long does the compilation take on your R-PI box? Are your compiling on an external USB SSD or on an SD card? We are compiling our images on an external SSD and the compilation process takes around 45 minutes.

davidk, 2021-03-15:
Hello Vojtěch!

It is great to hear that things are expanding in that direction for Prusa Research and PrusaSlicer. I would be happy to help further your efforts along, to the extent of my knowledge.

You are correct, all AppImages have been built on Raspberry Pi 4s. The main focus has been on simple reproducibility (in case this repository ever fell by the wayside it could be forked, built and maintained by any interested party with a Pi 4).

The current AppImage builds are done on two Pis:

- A 32-bit builder: RPi 4, 4GB with Raspberry Pi OS and USB3 attached Samsung SSD as storage.

- And a 64-bit builder: RPi 4, 8GB with [Raspberry Pi's 64-bit beta OS](https://www.raspberrypi.org/forums/viewtopic.php?t=275370) and µSD card for storage. Releases have also been built with Ubuntu Mate 20.10, which has more up-to-date dependencies (this required some workarounds located in the 20.10 branch of this repository: https://github.com/davidk/PrusaSlicer-ARM.AppImage/tree/20.10).

Cross-compiling was something I had been looking into most recently, since it aligned with another user's desire for a [containerized PrusaSlicer builder](https://github.com/davidk/docker-slic3r-prusa3d/issues/5). I believe it may have been possible to combine the work there with [docker's buildx plugin](https://docs.docker.com/buildx/working-with-buildx/) and build an AppImage for this project in unison, but I have not been able to look as closely, yet.

For build timing, using `time`, both RPi 4s clock in at 90/92 minutes of real time (32-bit (SSD)/64-bit (µSD card), respectively).

davidk, 2021-03-16:
Clarifying the above time a little bit: the build script is setup to build both GTK2 and GTK3 in one go, so that is the combined real time from start to AppImage generation (`time ./build.sh`) . 

Running `make -j$(nproc)` itself to build the PrusaSlicer binary takes 57m13.396s on the 32-bit builder (4GB, SSD) and 61m25.418s on the 64-bit builder (8GB, µSD).

bubnikv, 2021-03-16:
Thanks David for your response.  I hoped you are building both 32 and 64bit builds on the same 64bit R-PI4.

We will evaluate cross-compilation. We did that for the new Apple M1. We had difficulties to cross-compile dependencies on X86 for ARM, so we compiled the dependencies on our M1 laptop, but the builds of the application are cross-compiled.

We are getting another chromebook with Qualcomm ARM CPU, let's see how it will compare to the R-Pi, IMHO the virtualized MMC storage is a bummer. At the end of the day we may duplicate your setup with two R-PIs.

It surprised me a lot, that compilation on the external SSD takes the same time as on uSD. Our 32bit R-Pi build server compiles on external SSD as well.