davidk/PrusaSlicer-ARM.AppImage #16: Build .yml does not execute

theofficialgman, 2021-04-04:
running the ./build.sh to make a appimage for ubuntu bionic aarch64
The build script seems to not be working at all

I have all the dependencies installed but the .yml is just printed to the terminal
[output.txt](https://github.com/davidk/PrusaSlicer-ARM.AppImage/files/6253978/output.txt)

theofficialgman, 2021-04-04:
oh I see one of the problems
inside the function.sh the arm architecture is commented out

davidk, 2021-04-06:
Hello,

There is a workaround in the `20.10` branch (for Ubuntu 20.10, Groovy) which stubs the architecture presented to the packager. I believe that this might work in your situation as well: https://github.com/davidk/PrusaSlicer-ARM.AppImage/commit/dd6d6cc6f481529db13d6ef8ac76fe5601318462

The effective parts are:

- An executable file named `uname` that just echos 'unknown'
- A changed PATH envvar to `pkg2appimage` with the path to the `uname` file above in the first position: `PATH="${OLD_CWD}:${PATH}" SYSTEM_ARCH="${APPIMAGE_ARCH}" ./pkg2appimage ps.yml`

Bionic isn't on my radar at the moment, so i'm unable to provide more comprehensive assistance, but I hope this helps move your build forward.

theofficialgman, 2021-04-06:
ah well adding aarch64 to allowed architectures to both the  main functions.sh as well as the function.sh within prusaslicer allowed me to build (they only have x86, and i386 with arm commented out)

it seems to me that this would be an issue for building on all aarch64 and arm devices but let me know if you can build on your master branch with the latest prusaslicer commit without issues

davidk, 2021-04-07:
Builds work on Raspbian/Raspberry Pi 32-bit/64-bit; they are a bit behind the curve and report 'unknown' for `uname` which lets the build complete

theofficialgman, 2022-02-28:
closing as this issue is outdated

closed 2022-02-28 by theofficialgman