davidk/PrusaSlicer-ARM.AppImage #20: 2.4.0-alpha1 does not work with fedora 34 aarch64

xarbit, 2021-09-07:
I tried your AppImage builds for fedora 34 aarch64, both gtk3 and gtk2 fail to start. 

`prusa-slicer: error while loading shared libraries: libboost_log.so.1.67.0: cannot open shared object file: No such file or directory`

Package boost-log-1.75.0-4.fc34.aarch64 is currently in the repository.

davidk, 2021-09-09:
Hello!

It seems that things are quite different with respect to Fedora and how the current AppImages are packaged.. so i've built a version using Fedora 34's aarch64 distribution.

The resulting AppImage can be found here: https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/download/version_2.4.0-alpha1/PrusaSlicer-2.4.0-alpha1-GTK3-aarch64-Fedora-34.AppImage

These were the dependencies used at build time, if any are needed to run: `git cmake CGAL-devel boost-devel boost-regex boost-filesystem boost-thread boost-log boost-locale boost-iostreams libcurl pkgconf-pkg-config libbtbb cereal-devel eigen3-devel NLopt-devel systemd-devel openvdb-devel ImageMagick  file desktop-file-utils make automake gcc gcc-c++ tbb-devel libcurl-devel libtvdb-devel openvdb ilmbase-devel dpkg expat-devel glew-devel ninja-build openssl-devel perl-FindBin libXext-devel gtk2-devel gtk3-devel fuse`

xarbit, 2021-09-09:
wouldn't it be possible to create one AppImage for all distributions?

xarbit, 2021-09-09:
anyway, now I get: 

`./PrusaSlicer-2.4.0-alpha1-GTK3-aarch64-Fedora-34.AppImage 
prusa-slicer: error while loading shared libraries: libGLU.so.1: cannot open shared object file: No such file or directory
`

the AppImage doesn't seem to include the dependencies.

davidk, 2021-09-09:
I believe that Mesa-libGLU should bring that dependency in, apologies for the trouble

Tighter integration was in the works to make a more complete AppImage (dependencies as well), but Prusa Research appeared to be interested in producing these, so it did not seem like a good idea to continue development in parallel

I'll have another look at that if time allows it, here

closed 2021-12-07 by xarbit