davidk/PrusaSlicer-ARM.AppImage #21: Libboost linking directly to 1.67 (making these builds incompatible with anything but V

theofficialgman, 2021-11-08:
**Description of the issue**

**Hardware environment and OS that is being used to run PrusaSlicer-ARM.AppImage**

 - Hardware [e.g. Raspberry Pi 4 (4GB), Raspberry Pi 4 (8GB)]: 
 - OS [e.g. Raspberry Pi OS (32-bit/armhf), Ubuntu Mate 21.04 (64-bit/aarch64)]: 

**Steps to reproduce the reported issue (if applicable)**

**Screenshots (if applicable)**

Botspot, 2021-11-12:
I *almost* got it to work by creating a bunch of symlinks:
```bash
sudo apt-get install -y git cmake libboost-dev libboost-regex-dev libboost-filesystem-dev libboost-thread-dev libboost-log-dev libboost-locale-dev libcurl4-openssl-dev build-essential pkg-config libtbb-dev zlib1g-dev libcereal-dev libeigen3-dev libnlopt-cxx-dev libudev-dev libopenvdb-dev libboost-iostreams-dev libgmpxx4ldbl libnlopt-dev libdbus-1-dev imagemagick libgtk2.0-dev libgtk-3-dev libwxgtk3.0-gtk3-dev
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_log.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_log.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_filesystem.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_filesystem.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_thread.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_thread.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_log_setup.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_log_setup.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_locale.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_locale.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_chrono.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_chrono.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_atomic.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_atomic.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_date_time.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_date_time.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libopenvdb.so.7.1.0 /usr/lib/arm-linux-gnueabihf/libopenvdb.so.5.2
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_system.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_system.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_iostreams.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_iostreams.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.74.0 /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.67.0
sudo ln -s /usr/lib/arm-linux-gnueabihf/libHalf-2_5.so.25.0.3 /usr/lib/arm-linux-gnueabihf/libHalf.so.23
~/PrusaSlicer.AppImage
```
After all of that symlinking, PrusaSlicer no longer complains of missing libraries. But now it says:
```
prusa-slicer: symbol lookup error: prusa-slicer: undefined symbol: _ZN7openvdb4v5_24util11INVALID_IDXE
```
Other notes: The `libwxgtk3.0-dev` package is unavailable on Bullseye.

theofficialgman, 2021-11-12:
`libwxgtk3.0-dev` is the gtk2 development libraries, they aren't necessary if you use the gtk3 version of the appimage (which pi-apps does)
gtk2 has been discontinued in bullseye

Botspot, 2021-11-12:
> `libwxgtk3.0-dev` is the gtk2 development libraries, they aren't necessary if you use the gtk3 version of the appimage (which pi-apps does) gtk2 has been discontinued in bullseye

Are you sure that a package mentioning "gtk3.0" is only for gtk2?

theofficialgman, 2021-11-12:
> > `libwxgtk3.0-dev` is the gtk2 development libraries, they aren't necessary if you use the gtk3 version of the appimage (which pi-apps does) gtk2 has been discontinued in bullseye
> 
> Are you sure that a package mentioning "gtk3.0" is only for gtk2?

YUP it is. confusing as that might be
`libwxgtk3.0-dev` is for gtk2 and `libwxgtk3.0-gtk3-dev` is for gtk3
you can find that on the debian package repository

theofficialgman, 2021-11-12:
if you look at the files inside the `libwxgtk3.0-dev`
you find stuff like `/usr/lib/aarch64-linux-gnu/libwx_gtk2u_adv-3.0.so`
which points to it being gtk2

davidk, 2021-11-15:
Hello,

I have been working on bundling more dependencies into a single AppImage to work around this issue.

If you are able to, please give these a try; more dependencies have been bundled for compatibility (at the cost of increased size and download time).

- ARM64/aarch64: [PrusaSlicer-version_2.4.0-beta1-aarch64.appimagebuilder.AppImage](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/download/version_2.4.0-beta1/PrusaSlicer-version_2.4.0-beta1-aarch64.appimagebuilder.AppImage)

- armhf: [PrusaSlicer-version_2.4.0-beta1-armhf.appimagebuilder.AppImage](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/download/version_2.4.0-beta1/PrusaSlicer-version_2.4.0-beta1-armhf.appimagebuilder.AppImage)

Brief testing seems to show that these AppImages are backwards and forwards compatible with the distributions I have been able to use locally (Fedora, Raspberry Pi OS 10/11 64-bit and Ubuntu 21.10), but please let me know if something does not work and I will take a look

Botspot, 2021-11-15:
> [PrusaSlicer-version_2.4.0-beta1-armhf.appimagebuilder.AppImage](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/download/version_2.4.0-beta1/PrusaSlicer-version_2.4.0-beta1-armhf.appimagebuilder.AppImage)

Armhf appimage tested and working on my Bullseye system.

theofficialgman, 2021-11-15:
@davidk 
tested on ubuntu bionic arm64 and this above build ^ does NOT work
`[2021-11-15 18:16:31.880493] [0x0000007fac785010] [error]   Unable to init glew library`

theofficialgman, 2021-11-15:
also note I do have a working build of prusaslicer myself off of master https://github.com/cobalt2727/L4T-Megascript/blob/master/scripts/extras/prusaslicer.sh
^ this is what I need to build it. the current implementation of FindTBB.cmake is out of date in prusaslicer repo, so this script pulls in a working version (well, working for buster, bionic, focal, bullseye... I think latest arch needs more changed but thats out of scope)

theofficialgman, 2021-11-15:
I truncated the error:
```
S_PIXBUF (pixbuf)' failed
[2021-11-15 18:30:15.876704] [0x0000007f9a355010] [trace]   Initializing StaticPrintConfigs
(prusa-slicer:11997): Gtk-WARNING **: 18:30:16.115: GTK+ module /tmp/.mount_PrusaSo6EvTt//usr/lib/aarch64-linux-gnu/gtk-2.0/modules/libgail.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
(prusa-slicer:11997): Gtk-WARNING **: 18:30:16.385: GTK+ module /tmp/.mount_PrusaSo6EvTt//usr/lib/aarch64-linux-gnu/gtk-2.0/modules/libgail.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
(prusa-slicer:11997): Gtk-WARNING **: 18:30:16.435: GTK+ module /tmp/.mount_PrusaSo6EvTt//usr/lib/aarch64-linux-gnu/gtk-2.0/modules/libgail.so cannot be loaded.
GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported.
(prusa-slicer:11997): Gtk-CRITICAL **: 18:30:24.176: gtk_window_resize: assertion 'height > 0' failed
[2021-11-15 18:30:25.526329] [0x0000007f9a355010] [error]   Unable to init glew library

```
this appears to be the whole thing

theofficialgman, 2021-11-15:
@davidk 
If check the linker in hirsute with MY prusa-slicer build from bionic, the only linker issues are as follows:
```
	libboost_log.so.1.67.0 => not found
	libboost_filesystem.so.1.67.0 => not found
	libboost_thread.so.1.67.0 => not found
	libboost_locale.so.1.67.0 => not found
	libboost_chrono.so.1.67.0 => not found
	libopenvdb.so.5.0 => not found
	libboost_system.so.1.67.0 => not found
	libboost_iostreams.so.1.67.0 => not found
	libboost_regex.so.1.67.0 => not found
	libHalf.so.12 => not found
	libGLEW.so.2.0 => not found
```
you could probably drastically decrease the filesize of your appimage and target only including these libraries

everything after bionic uses `libGLEW.so.2.1` from the `libglew2.1` package, while bionic uses an older `libglew2.0` package and `libGLEW.so.2.0`

buster already moved to `libglew2.1` and doesn't use the `libglew2.0` package.
this is what causes your appimage to fail on bionic, your appimage requires system `libGLEW.so.2.1` to be present...

theofficialgman, 2021-11-16:
also failing for me on ubuntu hirsute arm64. I have to choose "yes" in the first image and then I get this error. and before you ask, yes `en_US.UTF-8` locales are working on my system just fine

[image]
[image]

theofficialgman, 2021-11-16:
@davidk also it should be noted... the official prusa-slicer binary from the appimage doesn't link to libboost or libGLEW AT ALL

maybe its worth looking at that appimage to see what they did there? clearly they did something differently in the build to not require libboost in the linking of the binary or in the appimage

davidk, 2021-11-17:
The builds above follow the static compilation method that PrusaSlicer uses for their builds, so the dependencies are mostly theirs (including boost, glew, their wx fork). There are lots of packaged runtime deps so that these will go (almost) everywhere, as an all-in-one AppImage truer to the format was requested in another issue.

I will be building additional AppImages akin to the smaller AppImages of yore (leaning more on distribution dependencies) and bundling problematic deps like boost, as these fuller 'kitchen sink' AppImages are not great for users with resource constraints.

Regarding the crashes hit, I installed and looked at Ubuntu 20.04 and 21.04:

* Ubuntu 21.04/Hirsute: I attempted to reproduce the issue on a fresh desktop install, and i'm unable to with the original AppImage above. It appears to start, load and slice OK. My best guess is that there is an interaction between the bundled locales and the local system.
 
[image]

* Ubuntu 20.04/Focal: glew is one of the deps that PrusaSlicer builds and packages, so bundling the building system's libgl1 helps it find the gl that it needs (otherwise it tries to use the system libgl and crashes). Unfortunately there is another interaction somewhere whereupon attempting to slice any object it crashes with a segmentation fault; i've had a glance at the core in gdb, and traced some of its runtime activity but nothing appears to stand out. 

  Probably another conflict, but I do not think it would be productive to continue to chase it down. The static version of PrusaSlicer also refuses to build on this version, breaking on the OpenSSL/libcurl deps. 

It's likely that leaning more on system dependencies will probably resolve both issues.

Thanks again for the reports and quick testing

theofficialgman, 2021-11-17:
yeah I get that I'm just surprised that doing a ldd on the official bin in the prusaslicer appimage doesn't even show boost or glew at all in the linker
```
squashfs-root/usr/bin/bin$ ldd ./prusa-slicer
        linux-vdso.so.1 (0x00007ffdf3feb000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2a2fd02000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2a2fb20000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f2a2fb15000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f2a2faf9000)
        libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f2a2fa71000)
        libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f2a2f934000)
        libXext.so.6 (0x00007f2a2f720000)
        libgtk-3.so.0 => /lib/x86_64-linux-gnu/libgtk-3.so.0 (0x00007f2a2ef6f000)
        libgdk-3.so.0 => /lib/x86_64-linux-gnu/libgdk-3.so.0 (0x00007f2a2ee6a000)
        libatk-1.0.so.0 => /lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007f2a2ee40000)
        libgio-2.0.so.0 => /lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007f2a2ec5f000)
        libpangocairo-1.0.so.0 => /lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007f2a2ec4d000)
        libgdk_pixbuf-2.0.so.0 => /lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007f2a2ec23000)
        libcairo-gobject.so.2 => /lib/x86_64-linux-gnu/libcairo-gobject.so.2 (0x00007f2a2ec17000)
        libpango-1.0.so.0 => /lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007f2a2ebc8000)
        libcairo.so.2 => /lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f2a2eaa5000)
        libgobject-2.0.so.0 => /lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f2a2ea45000)
        libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f2a2e91c000)
        libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f2a2e8e6000)
        libGLU.so.1 => /lib/x86_64-linux-gnu/libGLU.so.1 (0x00007f2a2e874000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2a2e851000)
        libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f2a2e800000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2a2e60e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f2a2fd23000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2a2e4bf000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2a2e4a2000)
        libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f2a2e3ea000)
        libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f2a2e3c0000)
        libgmodule-2.0.so.0 => /lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007f2a2e3ba000)
        libXi.so.6 (0x00007f2a2e1aa000)
        libXfixes.so.3 => /lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f2a2e1a0000)
        libatk-bridge-2.0.so.0 => /lib/x86_64-linux-gnu/libatk-bridge-2.0.so.0 (0x00007f2a2e169000)
        libepoxy.so.0 => /lib/x86_64-linux-gnu/libepoxy.so.0 (0x00007f2a2e036000)
        libfribidi.so.0 => /lib/x86_64-linux-gnu/libfribidi.so.0 (0x00007f2a2e019000)
        libpangoft2-1.0.so.0 => /lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007f2a2e000000)
        libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f2a2defb000)
        libfontconfig.so.1 => /lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f2a2deb2000)
        libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f2a2ddf3000)
        libXinerama.so.1 => /lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007f2a2ddee000)
        libXrandr.so.2 => /lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007f2a2dde1000)
        libXcursor.so.1 => /lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007f2a2ddd4000)
        libXcomposite.so.1 => /lib/x86_64-linux-gnu/libXcomposite.so.1 (0x00007f2a2ddcd000)
        libXdamage.so.1 => /lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f2a2ddc8000)
        libxkbcommon.so.0 => /lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007f2a2dd86000)
        libwayland-cursor.so.0 => /lib/x86_64-linux-gnu/libwayland-cursor.so.0 (0x00007f2a2dd7b000)
        libwayland-egl.so.1 => /lib/x86_64-linux-gnu/libwayland-egl.so.1 (0x00007f2a2dd76000)
        libwayland-client.so.0 => /lib/x86_64-linux-gnu/libwayland-client.so.0 (0x00007f2a2dd65000)
        libmount.so.1 => /lib/x86_64-linux-gnu/libmount.so.1 (0x00007f2a2dd03000)
        libselinux.so.1 => /lib/x86_64-linux-gnu/libselinux.so.1 (0x00007f2a2dcd8000)
        libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f2a2dcbc000)
        libthai.so.0 => /lib/x86_64-linux-gnu/libthai.so.0 (0x00007f2a2dcb1000)
        libpixman-1.so.0 => /lib/x86_64-linux-gnu/libpixman-1.so.0 (0x00007f2a2dc0a000)
        libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f2a2dbd0000)
        libxcb-shm.so.0 => /lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007f2a2dbcb000)
        libxcb-render.so.0 => /lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007f2a2dbbc000)
        libXrender.so.1 => /lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f2a2d9b2000)
        libffi.so.7 => /lib/x86_64-linux-gnu/libffi.so.7 (0x00007f2a2d9a6000)
        libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f2a2d933000)
        libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007f2a2d882000)
        libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f2a2d87c000)
        libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f2a2d874000)
        libatspi.so.0 => /lib/x86_64-linux-gnu/libatspi.so.0 (0x00007f2a2d83d000)
[... cut ...]
        libdatrie.so.1 => /lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007f2a2d6e6000)
        liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f2a2d6bb000)
        liblz4.so.1 => /lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f2a2d69a000)
        libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f2a2d57c000)
        libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f2a2d562000)
        libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f2a2d53f000)
 ```

Botspot, 2021-11-26:
Any updates on this?

davidk, 2021-11-29:
This should be resolved with the [beta2 release](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/tag/version_2.4.0-beta2), with the exception of the latest Fedora 35 aarch64 which has a conflicting glibc version

Botspot, 2021-11-29:
> This should be resolved with the [beta2 release](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/tag/version_2.4.0-beta2), with the exception of the latest Fedora 35 aarch64 which has a conflicting glibc version

Thanks! The download URLs [have been updated](https://github.com/Botspot/pi-apps/commit/853798573a19682f16d8db90bdbbac951d1d923c) in our raspberry pi app store, so this issue can be closed.

closed 2021-11-29 by davidk