davidk/PrusaSlicer-ARM.AppImage #22: no libwxgtk3.0-dev installation candidate on 64bit Raspberry PiOS

Hermann-SW, 2022-03-13:
**Description of the issue**
There is no installation candidate for "libwxgtk3.0-dev" on latest 64bit Raspberry PiOS, readme apt-get fails.

**Hardware environment and OS that is being used to run PrusaSlicer-ARM.AppImage**

 - Raspberry Pi >=3 (I use Pi400): 
 - latest 64bit Raspberry PiOS (Bullseye)
 
**Steps to reproduce the reported issue (if applicable)**
"sudo apt-get install ..." fails:

**Screenshots (if applicable)**
```
Reading state information... Done
Package libwxgtk3.0-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libwxgtk3.0-dev' has no installation candidate
pi@pi400-64:~ $ 
```

Fix is simply removing last package from the apt-get command, "PrusaSlicer-version_2.4.1-aarch64.AppImage" runs fine on 64bit PiOS then.

davidk, 2022-03-14:
Hello,

I've removed libwxgtk3.0-dev from the latest release instructions. Apologies for the trouble and thank you for using the AppImage!

closed 2022-03-14 by davidk

Hermann-SW, 2022-03-14:
Thanks, and thanks for the new appimage. I used 32bit 2.3.3 appimage for some time, but had to open vncviewer session into a 32bit PiOS since my Pi400 runs 64bit PiOS. Now I use the latest PrusaSlicer with freecad on my main computer !
[
https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-v2-4-1-released/#post-592573](
https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-v2-4-1-released/#post-592573
)
[image]