davidk/PrusaSlicer-ARM.AppImage #23: Initializing StaticPrintConfigs Segmentation fault

daizhuo, 2022-04-28:
**Description of the issue**
I can't run this appimage, it shows these info.
I actually don't know too much about how to use rasperry pi, I just an 3d fan.

I got these info:

./PrusaSlicer-version_2.4.2-aarch64.AppImage
[2022-04-28 11:44:10.056142] [0x0000007fa1682040] [trace]   Initializing StaticPrintConfigs
Segmentation fault

**Hardware environment and OS that is being used to run PrusaSlicer-ARM.AppImage**

 - Hardware Raspberry Pi 4 (4GB)
 - OS [Raspberry Pi OS  (64-bit/aarch64)]: 
 - PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian

davidk, 2022-04-29:
Hello,
Thank you for reporting the issue. I've tested the AppImage against a freshly imaged Raspberry Pi OS 11 and cannot seem to reproduce the issue.

Unfortunately the message is not descriptive enough to offer good advice as well. Could you try to move `/home/pi/.config/PrusaSlicer` away to another directory temporarily, re-run and see if that resolves the issue?

If you are willing to, installing and running `strace` against the AppImage might reveal the source of the issue too:

    $ sudo apt install strace -y && strace ./PrusaSlicer-version_2.4.2-aarch64.AppImage

davidk, 2022-07-10:
Hello,

With the release of the alpha build of 2.5.0, please consider trying [PrusaSlicer-2.5.0-alpha2 ARM AppImages](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/tag/version_2.5.0-alpha2) to see if resolves the issue.

I attempted to reproduce this issue under Raspberry Pi OS aarch64 with 2.5.0 but could not.

If the new alpha release does not resolve the issue, I would like to attempt to reproduce this, but would need extra details on what settings cause the issue

davidk, 2023-02-02:
Hello,

Seeing that several versions have passed the original issue has likely become stale (or triaged upstream). Will close this issue, but please re-open if the issue reoccurs on more recent versions and is caused by the AppImage packaging/build. Thank you again for reporting!

closed 2023-02-02 by davidk