davidk/PrusaSlicer-ARM.AppImage #24: Error in yml run

guy111a, 2022-06-28:
Hello

1st of all, thank you for this effort ! this is a great thing.

but i do have a problem.

Followed all steps and getting this error,

guy@fjord:~/Downloads/PrusaSlicer-ARM.AppImage $ appimage-builder --recipe AppImageBuilder-armhf-minimal.yml 
Traceback (most recent call last):
  File "/usr/local/bin/appimage-builder", line 8, in <module>
    sys.exit(__main__())
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/__main__.py", line 40, in __main__
    raw_recipe_data = recipe_loader.load(args.recipe)
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/recipe/loader.py", line 76, in load
    return yaml.load(recipe_file, Loader=self._loader)
  File "/usr/local/lib/python3.9/dist-packages/yaml/__init__.py", line 81, in load
    return loader.get_single_data()
  File "/usr/local/lib/python3.9/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 64, in compose_node
    if self.check_event(AliasEvent):
  File "/usr/local/lib/python3.9/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/local/lib/python3.9/dist-packages/yaml/parser.py", line 449, in parse_block_mapping_value
    if not self.check_token(KeyToken, ValueToken, BlockEndToken):
  File "/usr/local/lib/python3.9/dist-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/local/lib/python3.9/dist-packages/yaml/scanner.py", line 258, in fetch_more_tokens
    raise ScannerError("while scanning for the next token", None,
yaml.scanner.ScannerError: while scanning for the next token
found character '%' that cannot start any token
  in "AppImageBuilder-armhf-minimal.yml", line 48, column 14
guy@fjord:~/Downloads/PrusaSlicer-ARM.AppImage $ 

Please help.

The system is Pi 4 B 8 GB
os is:
guy@fjord:~/Downloads/PrusaSlicer-ARM.AppImage $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Thank you
Guy

davidk, 2022-06-28:
Hello,

The error message that caused it to exit early before finishing the build may be further up.. it's unclear what it might be as I could not replicate it locally building against version_2.4.2 with an aarch64 distribution. I've noticed in certain cases that the build needs to be run twice to complete on certain runs. Could you try again and see if that helps?

guy111a, 2022-06-29:
Thank you Sir for the replay. 

So, i run the script for 3 times in a row. same error.

- [download 97% complete]
-- [download 98% complete]
-- [download 99% complete]
-- [download 100% complete]
-- verifying file...
       file='/home/guy/Downloads/PrusaSlicer-ARM.AppImage/PrusaSlicer/deps/build/wxWidgets/v3.1.4-patched.zip'
-- SHA256 hash of
    /home/guy/Downloads/PrusaSlicer-ARM.AppImage/PrusaSlicer/deps/build/wxWidgets/v3.1.4-patched.zip
  does not match expected value
    expected: 'a1e145a083d173cf320c0bd8522c7ee5829052b49b68fe5268ac84f0c576b940'
      actual: 'e51493d071ca56f383955a86144b63b3c23c24565e235848dbe0f99c5f5fd57c'
-- Hash mismatch, removing...
-- Retry after 60 seconds (attempt #5) ...
-- Using src='https://github.com/prusa3d/wxWidgets/archive/refs/heads/v3.1.4-patched.zip'
-- verifying file...
       file='/home/guy/Downloads/PrusaSlicer-ARM.AppImage/PrusaSlicer/deps/build/wxWidgets/v3.1.4-patched.zip'
-- SHA256 hash of
    /home/guy/Downloads/PrusaSlicer-ARM.AppImage/PrusaSlicer/deps/build/wxWidgets/v3.1.4-patched.zip
  does not match expected value
    expected: 'a1e145a083d173cf320c0bd8522c7ee5829052b49b68fe5268ac84f0c576b940'
      actual: 'e51493d071ca56f383955a86144b63b3c23c24565e235848dbe0f99c5f5fd57c'
-- Hash mismatch, removing...
CMake Error at dep_wxWidgets-stamp/download-dep_wxWidgets.cmake:159 (message):
  Each download failed!

    
    

make[2]: *** [CMakeFiles/dep_wxWidgets.dir/build.make:110: dep_wxWidgets-prefix/src/dep_wxWidgets-stamp/dep_wxWidgets-download] Error 1
make[1]: *** [CMakeFiles/Makefile2:122: CMakeFiles/dep_wxWidgets.dir/all] Error 2
make: *** [Makefile:103: all] Error 2
Traceback (most recent call last):
  File "/usr/local/bin/appimage-builder", line 8, in <module>
    sys.exit(__main__())
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/__main__.py", line 50, in __main__
    invoker.execute(commands)
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/invoker.py", line 29, in execute
    command()
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/commands/run_script.py", line 82, in __call__
    raise RuntimeError("Script exited with code: %s" % _proc.returncode)
RuntimeError: Script exited with code: 2

Also there is another issue that is may be related.
In the documentation there is : 

 appimage-builder --recipe AppImageBuilder-aarch64.yml

But that does not work.

guy@duster:~/Downloads/PrusaSlicer-ARM.AppImage $  appimage-builder --recipe AppImageBuilder-aarch64.yml
Traceback (most recent call last):
  File "/usr/local/bin/appimage-builder", line 8, in <module>
    sys.exit(__main__())
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/__main__.py", line 40, in __main__
    raw_recipe_data = recipe_loader.load(args.recipe)
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/recipe/loader.py", line 75, in load
    with open(path) as recipe_file:
FileNotFoundError: [Errno 2] No such file or directory: 'AppImageBuilder-aarch64.yml'
guy@duster:~/Downloads/PrusaSlicer-ARM.AppImage $ 

Only thing that works is : 
ge $ appimage-builder --recipe AppImageBuilder
AppImageBuilder-aarch64-minimal.yml  AppImageBuilder-armhf-minimal.yml    AppImageBuilder.yml

appimage-builder --recipe AppImageBuilder.yml

[... 87 lines cut ...]

My system is a Pi 4B ( one of them is 4GB Ram and the other is 8GB Ram )

Image $ uname -m
aarch64

 $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
guy@duster:~/Downloads/PrusaSlicer-ARM.AppImage $ 

Also i have installed the dependencies and updated the OS to latest. 

Now i will try the prepared by Cura.

Thank you

Guy

guy111a, 2022-06-29:
Also i tried the assets and got the next:

( the 8gb ram machine )

guy@fjord:~/Downloads $ ls -la 
total 169412
drwxr-xr-x  4 guy guy     4096 Jun 29 13:38 .
drwxr-xr-x 22 guy guy     4096 Jun 28 12:41 ..
drwxr-xr-x  4 guy guy     4096 Apr 28 14:10 PrusaSlicer-2.4.2+linux-aarch64-GTK2-202204280955
-rw-r--r--  1 guy guy 45442933 Jun 29 13:31 PrusaSlicer-2.4.2+linux-aarch64-GTK2-202204280955.tar.bz2
drwxr-xr-x  4 guy guy     4096 Apr 28 15:30 PrusaSlicer-2.4.2+linux-aarch64-GTK3-202204281113
-rw-r--r--  1 guy guy 45453744 Jun 29 12:50 PrusaSlicer-2.4.2+linux-aarch64-GTK3-202204281113.tar.bz2
-rwxr-xr-x  1 guy guy 82552000 Jun 29 13:16 PrusaSlicer-version_2.4.2-aarch64.AppImage
guy@fjord:~/Downloads $ 
guy@fjord:~/Downloads $ 
guy@fjord:~/Downloads $ ./PrusaSlicer-version_2.4.2-aarch64.AppImage 
[2022-06-29 13:42:36.059227] [0x0000007f874c3040] [trace]   Initializing StaticPrintConfigs
Segmentation fault
guy@fjord:~/Downloads $ PrusaSlicer-2.4.2+linux-aarch64-GTK2-202204280955
PrusaSlicer-2.4.2+linux-aarch64-GTK2-202204280955
guy@fjord:~/Downloads $ PrusaSlicer-2.4.2+linux-aarch64-GTK2-202204280955/prusa-slicer 
[2022-06-29 13:42:59.852235] [0x0000007fab6c7040] [trace]   Initializing StaticPrintConfigs
Segmentation fault  
guy@fjord:~/Downloads $ PrusaSlicer-2.4.2+linux-aarch64-GTK3-202204281113/prusa-slicer 
[2022-06-29 13:43:19.382895] [0x0000007fb39e5040] [trace]   Initializing StaticPrintConfigs
Segmentation fault
guy@fjord:~/Downloads $ 

I wander what is missing in my OS.

guy@fjord:~ $ uname -a
Linux fjord 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
guy@fjord:~ $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Would really like to get it to work.

If any other data is needed to understand the issue and fix, i would be happy to give 

Thank you

Guy

davidk, 2022-06-29:
Apologies. It looks like this is due to out of date and incorrect documentation. Running the build using `AppImageBuilder.yml` points at `version_2.4.0-beta1`, so dependency checksums would be out of date and the build would fail understandably.

Please remove the PrusaSlicer directory and run `build.sh` which should figure out the latest version of PrusaSlicer, choose the right .yml file to replace the version string of (this is the other errors you are seeing) and run the build.

Regarding the releases, the usual cause of a segmentation fault is running without `$DISPLAY` set (this happens over SSH) or missing libraries -- the dependencies should be listed in the release message.

guy111a, 2022-06-30:
Hello again, did as you wrote. and got this:

      |                     ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/libslic3r/Fill/FillAdaptive.cpp: In function ‘Slic3r::FillAdaptive::rtree_segment_t Slic3r::FillAdaptive::mk_rtree_seg(const Slic3r::Line&)’:
../src/libslic3r/Fill/FillAdaptive.cpp:664:31: note: parameter passing for argument of type ‘boost::geometry::model::segment<boost::geometry::model::point<float, 2, boost::geometry::cs::cartesian> >’ when C++17 is enabled changed to match C++14 in GCC 10.1
  664 | static inline rtree_segment_t mk_rtree_seg(const Line &l) {
      |                               ^~~~~~~~~~~~
../src/libslic3r/Fill/FillAdaptive.cpp: In function ‘std::pair<double, double> Slic3r::FillAdaptive::adaptive_fill_line_spacing(const Slic3r::PrintObject&)’:
../src/libslic3r/Fill/FillAdaptive.cpp:276:27: note: parameter passing for argument of type ‘std::pair<double, double>’ when C++17 is enabled changed to match C++14 in GCC 10.1
  276 | std::pair<double, double> adaptive_fill_line_spacing(const PrintObject &print_object)
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/usr/local/bin/appimage-builder", line 8, in <module>
    sys.exit(__main__())
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/__main__.py", line 50, in __main__
    invoker.execute(commands)
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/invoker.py", line 29, in execute
    command()
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/commands/run_script.py", line 82, in __call__
    raise RuntimeError("Script exited with code: %s" % _proc.returncode)
RuntimeError: Script exited with code: 1
Finished build process.
Here's some information to help with generating and posting a release on GitHub:
Tag: version_2.5.0-alpha2
-----
_Prusa Research has officially begun releasing their own [AppImages for ARM](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha2). It is advisable to switch to these official builds going forward._
This release mirrors PrusaSlicer's [upstream version_2.5.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha2). AppImages are now built using appimage-builder (with PrusaSlicer's dependencies) for broader compatibility at the cost of an increased AppImage size.
-----
Not sure about the next step. run this:

guy@fjord:~/Downloads/PrusaSlicer-ARM.AppImage $ ls -la 
total 240
drwxr-xr-x  5 guy guy   4096 Jun 30 05:06 .
drwxr-xr-x  3 guy guy   4096 Jun 30 01:33 ..
-rw-r--r--  1 guy guy   3240 Jun 30 01:33 AppImageBuilder-aarch64-minimal.yml
-rw-r--r--  1 guy guy   3227 Jun 30 01:33 AppImageBuilder-armhf-minimal.yml
-rw-r--r--  1 guy guy   3447 Jun 30 01:33 AppImageBuilder.yml
-rwxr-xr-x  1 guy guy   5216 Jun 30 01:33 build.sh
-rw-r--r--  1 guy guy    355 Jun 30 01:33 .editorconfig
drwxr-xr-x  8 guy guy   4096 Jun 30 01:33 .git
drwxr-xr-x  4 guy guy   4096 Jun 30 01:33 .github
-rw-r--r--  1 guy guy     56 Jun 30 01:33 .gitignore
-rw-r--r--  1 guy guy   1210 Jun 30 01:33 LICENSE
drwxr-xr-x 15 guy guy   4096 Jun 30 03:01 PrusaSlicer
-rw-r--r--  1 guy guy 186547 Jun 30 01:33 prusaslicer-on-pi.png
-rw-r--r--  1 guy guy   2164 Jun 30 01:33 README.md
guy@fjord:~/Downloads/PrusaSlicer-ARM.AppImage $ appimage-builder --recipe AppImageBuilder-aarch64-minimal.yml 
Traceback (most recent call last):
  File "/usr/local/bin/appimage-builder", line 8, in <module>
    sys.exit(__main__())
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/__main__.py", line 40, in __main__
    raw_recipe_data = recipe_loader.load(args.recipe)
  File "/usr/local/lib/python3.9/dist-packages/appimagebuilder/recipe/loader.py", line 76, in load
    return yaml.load(recipe_file, Loader=self._loader)
  File "/usr/local/lib/python3.9/dist-packages/yaml/__init__.py", line 81, in load
    return loader.get_single_data()
  File "/usr/local/lib/python3.9/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python3.9/dist-packages/yaml/composer.py", line 64, in compose_node
    if self.check_event(AliasEvent):
  File "/usr/local/lib/python3.9/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
[... cut ...]
  File "/usr/local/lib/python3.9/dist-packages/yaml/scanner.py", line 258, in fetch_more_tokens
    raise ScannerError("while scanning for the next token", None,
yaml.scanner.ScannerError: while scanning for the next token
found character '%' that cannot start any token
  in "AppImageBuilder-aarch64-minimal.yml", line 49, column 14
guy@fjord:~/Downloads/PrusaSlicer-ARM.AppImage $ 

* regarding the segmentation fault, was running on local host ( crowPi L )
* Not ssh or any other remote access.

Thank you

guy111a, 2022-06-30:
Tried again from scratch .. 
Failed here:

../deps/build/destdir/usr/local/include/CGAL/mpq_class.h: In member function ‘std::pair<double, double> CGAL::Real_embeddable_traits<__gmp_expr<__mpq_struct [1], __mpq_struct [1]> >::To_interval::operator()(const mpq_class&) const’:
../deps/build/destdir/usr/local/include/CGAL/mpq_class.h:204:48: note: parameter passing for argument of type ‘std::pair<double, double>’ when C++17 is enabled changed to match C++14 in GCC 10.1
  204 |         operator()( const mpq_class& x ) const {
      |                                                ^
In file included from ../deps/build/destdir/usr/local/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from ../deps/build/destdir/usr/local/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from ../deps/build/destdir/usr/local/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from ../deps/build/destdir/usr/local/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from ../deps/build/destdir/usr/local/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from ../deps/build/destdir/usr/local/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from ../deps/build/destdir/usr/local/include/boost/shared_ptr.hpp:17,
                 from ../deps/build/destdir/usr/local/include/boost/format/alt_sstream.hpp:22,
                 from ../deps/build/destdir/usr/local/include/boost/format/internals.hpp:24,
                 from ../deps/build/destdir/usr/local/include/boost/format.hpp:38,
                 from ../src/libslic3r/Semver.hpp:9,
                 from ../src/libslic3r/libslic3r.h:34,
                 from ../src/libslic3r/TriangleMesh.hpp:4,
                 from ../src/libslic3r/MeshBoolean.hpp:7,
                 from ../src/libslic3r/MeshBoolean.cpp:2:
../deps/build/destdir/usr/local/include/boost/bind.hpp: At global scope:
../deps/build/destdir/usr/local/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
../deps/build/destdir/usr/local/include/boost/function_output_iterator.hpp:14:1: note: ‘#pragma message: This header is deprecated. Use <boost/iterator/function_output_iterator.hpp> instead.’
   14 | BOOST_HEADER_DEPRECATED("<boost/iterator/function_output_iterator.hpp>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
../deps/build/destdir/usr/local/include/boost/detail/iterator.hpp:13:1: note: ‘#pragma message: This header is deprecated. Use <iterator> instead.’
   13 | BOOST_HEADER_DEPRECATED("<iterator>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.

Still no luck

Thank you

Guy

davidk, 2022-06-30:
Hello,

Thank you for your efforts. It is not always easy to figure out what is wrong this since builds can take quite a long time. 

If you are after the latest build, PrusaSlicer 2.5.0-alpha2 images are available [here](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/tag/version_2.5.0-alpha2).

---

```
ninja: build stopped: subcommand failed.
```

It's not clear why this fails.. If building for armhf then upstream PrusaSlicer had an issue where a test stopped the build; performing a `git pull` and rebuilding should let it complete.

```
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
```

Does the build system have enough memory? 8GB of memory is needed, but Pis with 4GB of memory need additional swap to complete (but this is very very slow and not recommended).

```
guy@fjord:/Downloads/PrusaSlicer-ARM.AppImage $ appimage-builder --recipe AppImageBuilder-aarch64-minimal.yml
Traceback (most recent call last):
File "/usr/local/bin/appimage-builder", line 8, in
sys.exit(main())
```

To run directly `AppImageBuilder-aarch64-minimal.yml` needs to be modified first: 

Change `%%VERSION%%` to a PrusaSlicer release (like `version_2.5.0-alpha2`) on:

Line 19
https://github.com/davidk/PrusaSlicer-ARM.AppImage/blob/dd46663c12adc7f976d7ff5f9cec659a4719c0e0/AppImageBuilder-aarch64-minimal.yml#L19 

And line 49 https://github.com/davidk/PrusaSlicer-ARM.AppImage/blob/dd46663c12adc7f976d7ff5f9cec659a4719c0e0/AppImageBuilder-aarch64-minimal.yml#L49

`build.sh` will handle this automatically, selecting the right build file based on the running system (armhf vs aarch64) along with building the current version of PrusaSlicer.

guy111a, 2022-07-01:
Hello

Thank you.
I technically do not need the latest or the compilation.. just need slicer
working .. and cannot get it to work...

Thank you

Guy

davidk, 2022-07-01:
Hello,

I imaged a new microSD card with the latest 64-bit Raspberry Pi OS `2022-04-04-raspios-bullseye-arm64.img.xz` and downloaded the latest aarch64 AppImage to test: [PrusaSlicer-version_2.5.0-alpha2-aarch64.AppImage](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/download/version_2.5.0-alpha2/PrusaSlicer-version_2.5.0-alpha2-aarch64.AppImage). 

[image]

It did not require any dependencies to start and slice on a Pi 400, so I suspect the issue is likely environment or OS related.

If there is a .PrusaSlicer directory in `~/.config`, please try moving it away and running. This can sometimes cause crashes before starting. 

Installing and running `strace` against the AppImage may also give some hints as to what is happening: `strace ./PrusaSlicer-version_2.5.0-alpha2-aarch64.AppImage`.

guy111a, 2022-07-01:
Thank you very much for your effort!

I use the same os version as you

I will try now to do as you suggested

Thank you

Guy

guy111a, 2022-07-02:
Thank you David for last note

After deleting the caches. The appImage worked fine and i am able to send a
print !

But :) while searching for a screenshot tool to show you i succeeded .. i
came across a tool called pi-apps .. thamuch ables a seemless install of
cura 😉

So now i have both running

Again.. Thank you very much for your effort and help.

Highly appreciated

Guy

davidk, 2022-07-05:
Hello,

That's great news that it's working now :tada:. Thanks for working on this until it was resolved, please re-open if there are any other issues caused by the build.

closed 2022-07-05 by davidk