davidk/PrusaSlicer-ARM.AppImage #26: 2.60 alpha 2

ericd1949, 2023-02-01:
Hi David,
Prusa are having difficulties with the rpi release for 2.60 alpha 2, so they are probably going to leave it till the next release.
Quote:
"We are having technical issues rolling out the RPi build and decided not to hinder the alpha release. Depending on when we manage to fix them, we will either attach them to 2.6.0-alpha2 or include them with the follow-up releases."

You could be a hero!!  If you've got the time, of course. :-)
Cheers,
Eric

ericd1949, 2023-02-01:
Brilliant David, it works a treat, thanks very much.
Cheers,
Eric

closed 2023-02-01 by ericd1949

davidk, 2023-02-02:
I imagine it's something they'll fix pretty quickly.. so please do consider using the officially/professionally supported versions if and when they do pop up. Until then, happy to provide a build to bridge the gap! Cheers and best of luck with the alpha!