davidk/PrusaSlicer-ARM.AppImage #28: PrusaSlicer-version_2.6.0-alpha6-aarch64.AppImage doesn' t start

WilliamHJKL, 2023-04-16:
**Description of the issue**
- PrusaSlicer doesn't start, fails with errors related to pixbuf.
- 
**Hardware environment and OS that is being used to run PrusaSlicer-ARM.AppImage**
- Raspberry Pi 4 (8GB)
- OS = Debian GNU/Linux 11 (bullseye)
- Linux pi4 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
- Fresh install of Raspbian Lite. + xserver / xinit / i3 etc ...

**Steps to reproduce the reported issue (if applicable)**
- Install the dependencies as indicated in the doc.
- Download the appimage
- chmod +x then 
./PrusaSlicer-version_2.6.0-alpha6-aarch64.AppImage

[2023-04-16 18:00:00.641626] [0x0000007fac61d040] [trace]   Initializing StaticPrintConfigs

(prusa-slicer:16888): Gtk-WARNING **: 18:00:05.473: Could not find the icon 'dialog-question'. The 'hicolor' theme
was not found either, perhaps you need to install it.
You can get a copy from:
	http://icon-theme.freedesktop.org/releases

(prusa-slicer:16888): Gtk-WARNING **: 18:00:11.282: Could not load a pixbuf from /org/gtk/libgtk/icons/16x16/status/image-missing.png.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognised image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognised image file format (gdk-pixbuf-error-quark, 3)
Aborted

hicolor-icon-theme is installed on the system.
Tested with 2.5.1 and 2.5.2 minimal AppImage, same results, same error messages.

davidk, 2023-04-16:
Hello,
Thank you for the information! The AppImages are tested against the full desktop version of Raspbian/Raspberry Pi OS.. so there may be additional setup steps missing on Lite even after conversion to an i3 desktop.

My best guess from the error message is that the libpng loader is not being found. Please try running the following to see if it fixes the issue:

```bash
sudo update-mime-database /usr/share/mime
sudo /usr/lib/arm-linux-gnueabihf/gdk-pixbuf-2.0/gdk-pixbuf-query-loaders --update-cache
```

Prepending `strace -e trace=openat` should give more information if more dependencies are missing.

WilliamHJKL, 2023-04-16:
Hello,

No luck after trying running 
- sudo update-mime-database /usr/share/mime
- sudo /usr/lib/aarch64-linux-gnu/gdk-pixbuf-2.0/gdk-pixbuf-query-loaders --update-cache
I also suspect a missing dependency, but I can't see which one with strace. (attaching the log of the command if it could help)

[prusa.log](https://github.com/davidk/PrusaSlicer-ARM.AppImage/files/11243271/prusa.log)

Thanks.

davidk, 2023-04-17:
Thanks for the log, I brought up a fresh install of Raspberry Pi OS Lite 64-bit to debug and further reproduce the issue.. I'm able to start the GUI with the following line: `XDG_DATA_DIRS="/usr/share" ./PrusaSlicer-version_2.5.2-aarch64.AppImage`

Is `XDG_DATA_DIRS` set to something appropriate? If not please try the line above to see if that brings up the GUI -- this is normally set in the full distribution, but was not when going from a Lite to i3 desktop

WilliamHJKL, 2023-04-17:
XDG_DATA_DIRS is not set ... Defining it before running PrusaSlicer solves the problem. Thx.
I just need to find which package|script defines it in the full distribution....

davidk, 2024-07-05:
Hello,
Closing this issue as resolved, please re-open if needed.

closed 2024-07-05 by davidk