davidk/PrusaSlicer-ARM.AppImage #29: Application segfaults on start on piOS Bookworm ARM64

theofficialgman, 2023-10-14:
**Description of the issue**
Core dumps immediately on running the appimage.

**Hardware environment and OS that is being used to run PrusaSlicer-ARM.AppImage**

 - Hardware: Raspberry Pi 5 8GB
 - OS: PiOS Bookworm ARM64

davidk, 2023-10-15:
I was not able to reproduce a core with Bookworm on an RPi 4. Could you provide more details of the core and any other errors you may have received, as I do not have a RPi 5.

On the distribution side, Bookworm appears to be setting the initial locale on some installs to [non-UTF8](https://forums.raspberrypi.com/viewtopic.php?t=357707), which may result in it exiting early.

The workaround for this is to enter and set `Preferences > Raspberry Pi Configuration > Localization > Set Locale > Character Set > UTF-8`

[image]
[image]

PrusaSlicer itself should probably not hard exit when encountering such a situation, but it looks like issue 11009 is open upstream for this.

theofficialgman, 2023-10-30:
> On the distribution side, Bookworm appears to be setting the initial locale on some installs to [non-UTF8](https://forums.raspberrypi.com/viewtopic.php?t=357707), which may result in it exiting [...]

I reported it upstream to PiOS and they fixed their piwiz GUI now https://github.com/raspberrypi-ui/piwiz/commit/4a801cbb092784653d8958dbf6c8d2b0a96500d8 . Should be fixed when the next image release gets made. For users now, they will have to use rpiconfig as you mention.

> PrusaSlicer itself should probably not hard exit when encountering such a situation, but it looks like issue 11009 is open upstream for this.

I get that popup and then it coredumps. Sorry I should have written that in the initial post. Will try with the locale corrected later.

theofficialgman, 2023-11-03:
Sorry it took me so long to test.
With the locale character set corrected it does work. Hopefully rpi ltd gets a new image out soon so users don't run into this.

closed 2023-11-03 by theofficialgman