davidk/PrusaSlicer-ARM.AppImage #31: No such File Error

AldoBertD, 2024-03-17:
Hello David,

I really appreciate your work. I am trying to download PrusaSlicer for a Raspberry Pi 4 Model B with Raspbian. I've downloaded the full repository and ran ./build.sh in the terminal. Everything runs fine until it prompts me to choose between the full version, the minimum version, or both. When I execute any command at that point, it gives me the following message:

FileNotFoundError: [Errno 2] No such file or directory: 'AppImageBuilder-armhf-full-version_2.7.2.yml'

I tried changing the names of the yml files but I see that there is no improvement

davidk, 2024-03-18:
Hey there @AldoBertD - that error is likely due to the dependencies being different than expected. I would suggest using the container build environments for consistency: https://github.com/davidk/PrusaSlicer-ARM.AppImage/tree/e24de209ff397566a17c6e5cdeb42dc757874b76/containers

If desired, the pre-built AppImages are here (these are the outputs of the container/k8s utilities): https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/tag/version_2.7.2

AldoBertD, 2024-03-19:
Didn't work as expected, i downloaded the AppImage from the page of prusa, but once y try to generate de gcode it simply closed the app with no message, did you manage to install and use correctly prusa slicer in the raspberry pi 4 model B 8 gb ram, cause i really need to download it in the raspberry.

davidk, 2024-03-20:
There seems an issue related to the OpenGL version with the latest 2.7.2 release from PrusaSlicer. Please try the `-opengl.AppImage` files in: https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/tag/version_2.7.2 to see if it resolves the issue.

I have tested it under Raspberry Pi OS Bookworm 32-bit and I am able to load, slice and export g-code.

AldoBertD, 2024-03-23:
Still cannot dowload it correctly, is there any video of the process on how to download it with your ARM? i really needed it

davidk, 2024-03-23:
There is no video walk-through and I don't recall if anyone has created one for this particular AppImage. I have attached an image of the installation process on Bookworm. The process is roughly:

```
$ sudo apt install libfuse2
$ dpkg --print-architecture 
$ chmod +x ./PrusaSlicer-version_2.7.2-minimal-armhf-opengl3.AppImage
$ ./PrusaSlicer-version_2.7.2-minimal-armhf-opengl3.AppImage
```
[image]

If that does not work, please provide some details about your system, the errors you are seeing, and I will see what I can do to replicate your issue. 

Please also consider using [Pi-Apps](https://pi-apps.io/) - they provide a more user-friendly way to use this AppImage too.

AldoBertD, 2024-03-25:
"Already I've done everything you mention, I downloaded Pi-Apps and ran all the mentioned command lines. However, I'm getting a 'Violacion de Segmento' message, which means 'segmentation fault' in English. It won't let me run it even using sudo beforehand, nor opening it in the same folder and running it."
[image]

davidk, 2024-03-25:
It looks like the AppImage build is too new for OSes below Bookworm. Check with the following:

```bash
pi@raspberrypi:~/Downloads $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

The way that PrusaSlicer was built, it should still work on Bullseye if the files are unpacked. Without upgrading the entire OS, the following should work:
```
$ ./PrusaSlicer-version_2.7.2-minimal-armhf-opengl3.AppImage --appimage-extract
$ cd squashfs-root/
$ MESA_GL_VERSION_OVERRIDE=3.2 ./usr/bin/prusa-slicer
```

AldoBertD, 2024-03-26:
I just download the new Raspberry Pi OS Bookworm 64-bit i just did the same process over again with all your recommendations. i used the aarch64 version and didnt work again, i can open the app but once i click sliced it closes with no info about it, i think your AppBuilder is optimized for 32-bit, let me know if you have test it with Raspberry Pi OS Bookworm 64-bit please

davidk, 2024-03-26:
[image]

I am able to slice and export from a fresh installation using:

```bash
    7  sudo apt install -y libfuse2
    8  chmod +x ./PrusaSlicer-version_2.7.2-aarch64-opengl3.AppImage 
    9  ./PrusaSlicer-version_2.7.2-aarch64-opengl3.AppImage 
```

```
$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

If you see any error messages in the terminal, it would be helpful in pinpointing the direct cause

AldoBertD, 2024-03-29:
I truly appreciate all the time you've dedicated to this matter. I was finally able to resolve the issue after installing the latest operating system multiple times on the Raspberry Pi 4 Model B. Ultimately, the error was traced back to the Spanish version. Once the English version was installed, PrusaSlicer ran correctly as mentioned in your last comment. I still don't understand why the language of the operating system would affect the use of the program. Once again, thank you very much for your help. I truly appreciate it.

closed 2024-03-29 by AldoBertD

AldoBertD, 2024-03-29:
Just tested it out again, its seems that also installing vnc-server also influences the use of the PrusaSlicer ARM, this vnc-server was the app i use to visualize the raspberry pi via Wifi on my computer.