davidk/PrusaSlicer-ARM.AppImage #32: No GUI build

andiwinter, 2024-04-06:
**Describe the feature and if there are any other alternatives (if applicable)**
It would be nice to have also a no GUI build if you want to use the slicing feature only

**Does this feature resolve a problem?**
You would be able to slice files also if your embedded system has no display at all

**Additional context (screenshots, diagrams, links to external documentation etc)**
https://github.com/prusa3d/PrusaSlicer/issues/7521 should explain how it can be done. I tried to compile it on my Raspberry Pi 3 with 32-bit Bullseye installed, but the compilation process always got stuck.

davidk, 2024-04-06:
Hi there,
PrusaSlicer (and this 3rd party AppImage) ships with the ability to use no GUI, but it is not primarily supported upstream as the linked issue shows. While I've used this functionality in the [past for experiments](https://github.com/davidk/slic3r-action-test), I think it would be best to stay as close to upstream offerings as possible, continuing to offer both the combined GUI+CLI in one.

If it helps, this shows the current AppImage with the CLI being used for slicing, along with the gcodeviewer showing the result:

[image]

andiwinter, 2024-04-07:
Thanks for the clarification, I didn't expect that it's a combined image executing either CLI or GUI depending on the parameters called with.

closed 2024-04-07 by andiwinter