davidk/PrusaSlicer-ARM.AppImage #33: Raspberry Pi5-New Prusa with Pi Apps not loading

APShark, 2024-07-02:
Using Pi APPS on a RP5 with 8GB ram- 500GB NVMe drive, Raspberry Pi OS 64 Bit.
I upgraded from previous version 27.4 I believe. That ran just fine.
Went to run the from DESKTOP MENU the program icon, and nothing came up.
Tried to run from TERMINAL  and got this:
MESA_GL_VERSION_OVERRIDE=3.3 /opt/PrusaSlicer.AppImage
/tmp/.mount_PrusaSyXJF61/usr/bin/prusa-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

Ok, please let me know if you have an answer for this, and thanks for producing the app images for the Pi.

davidk, 2024-07-03:
Hello,
The version of `libwebkit2gtk` is OS dependent and could not be packaged with the AppImage for this release, similar to what has happened with the upstream PrusaSlicer release. 

The library PrusaSlicer needs is provided by the following package on bookworm:

```bash
pi@raspberrypi:~ $ dpkg -S libwebkit2gtk-4.0.so.37
libwebkit2gtk-4.0-37:arm64: /usr/lib/aarch64-linux-gnu/libwebkit2gtk-4.0.so.37
libwebkit2gtk-4.0-37:arm64: /usr/lib/aarch64-linux-gnu/libwebkit2gtk-4.0.so.37.68.6
```

Please run the following to install, then re-run PrusaSlicer to see if it resolves the issue:

```bash
pi@raspberrypi:~ $ sudo apt-get install libwebkit2gtk-4.0-37
```

APShark, 2024-07-03:
Yes, working perfectly now, thank you!Sent from my iPhone 10XRWith Kind Regards,Mr. Albert P.  Sharkis Sr.Industrial Controls Support SpecialistKingman, AZ  USA

davidk, 2024-07-05:
Hello,
Closing this issue, thanks for confirming that it worked!

closed 2024-07-05 by davidk

APShark, 2024-07-05:
Yes, the solution was perfect!Thank you for your quick response. Sent from my iPhone 10XRWith Kind Regards,Mr. Albert P.  Sharkis Sr.Industrial Controls Support SpecialistKingman, AZ  USA