davidk/PrusaSlicer-ARM.AppImage #34: Error when Running: GLIBC_xx not found

HeavyFalcon678, 2024-09-24:
When I ran this in the GUI it did not work, so I tried the terminal and this was the error code. 

`/tmp/.mount_PrusaS1n0nzr/usr/bin/prusa-slicer: /lib/aarch64-linux-gnu/libm.so.6: version 'GLIBC_2.38' not found (required by /tmp/.mount_PrusaS1n0nzr/usr/bin/prusa-slicer)
/tmp/.mount_PrusaS1n0nzr/usr/bin/prusa-slicer: /lib/aarch64-linux-gnu/libstdc++.so.6: version 'GLIBCXX_3.4.32' not found (required by /tmp/.mount_PrusaS1n0nzr/usr/bin/prusa-slicer)
/tmp/.mount_PrusaS1n0nzr/usr/bin/prusa-slicer: /lib/aarch64-linux-gnu/libc.so.6: version 'GLIBC_2.38' not found (required by /tmp/.mount_PrusaS1n0nzr/usr/bin/prusa-slicer)
`

I am running the standard Raspberry Pi OS (Bookworm) in 64 bit.

davidk, 2024-09-24:
Hello,
Appreciate the report, it looks like the GLIBC version on the build platform is a bit too recent. I've updated all the AppImages in the 2.8.1 release with one that should work with Bookworm.

Please [pull down the latest](https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/tag/version_2.8.1) images, these should resolve the issue

HeavyFalcon678, 2024-09-24:
Thank you for your prompt response. The GLIBC error has been fixed. But now when I run the most recent AppImage (PrusaSlicer-version_2.8.1-aarch64-latestOS.AppImage) it shows this error: 

```
(prusa-slicer:66720): Gtk-WARNING **: 10:35:30.686: Could not load a pixbuf from /org/gtk/libgtk/icons/16x16/actions/dialog-question.png.
This may indicate that pixbuf loaders or the mime database could not be found.

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:39.876: gtk_window_resize: assertion 'width > 0' failed

(prusa-slicer:66720): GLib-GObject-WARNING **: 10:35:40.025: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.026: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:66720): GLib-GObject-WARNING **: 10:35:40.031: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.031: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:66720): GLib-GObject-WARNING **: 10:35:40.048: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.048: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:66720): GLib-GObject-WARNING **: 10:35:40.053: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.053: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:66720): GLib-GObject-WARNING **: 10:35:40.082: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.082: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:66720): GLib-GObject-WARNING **: 10:35:40.087: invalid cast from 'wxPizza' to 'GtkCellLayout'

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.088: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.283: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.283: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.285: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.306: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.306: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:66720): Gtk-CRITICAL **: 10:35:40.307: gtk_widget_set_size_request: assertion 'width >= -1' failed
[2024-09-24 10:35:40.894267] [0x0000007fa902a040] [error]   UserAccount: Failed to read token - no datafile found.

(prusa-slicer:66720): Gtk-WARNING **: 10:35:41.105: Could not find the icon 'user-home-symbolic-ltr'. The 'hicolor' theme
was not found either, perhaps you need to install it.
You can get a copy from:
	http://icon-theme.freedesktop.org/releases
**
Gtk:ERROR:../../../gtk/gtkiconhelper.c:495:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../gtk/gtkiconhelper.c:495:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Aborted
```

Thank you again for your support and for making this software available to the Raspberry Pi community.

davidk, 2024-09-25:
Thanks for following up, and am glad to hear that the GLIBC issue was resolved!

For the new issue; this has popped up in the past and may be related to the default fallback theme not being found.

Please try the following:

- `sudo apt-get install hicolor-icon-theme`
- `sudo gtk-update-icon-cache /usr/share/icons/hicolor`
- Check that `XDG_DATA_DIRS` is set as well: `echo $XDG_DATA_DIRS`

I am unable to reproduce the issue on a new installation of Raspberry Pi OS, but hopefully these will get the AppImage working or narrow down the issue a bit further

[image]

HeavyFalcon678, 2024-09-26:
Thank you very much. That fixed the issue.

closed 2024-09-26 by HeavyFalcon678