davidk/PrusaSlicer-ARM.AppImage #35: Prusa login fails

fberst, 2024-10-24:
When i try to log into my Prusa account the login pages is blank and shows only "Connection failed, something went wrong"

I use the current Raspberry Pi OS 64bit (based an Debian 12 bookworm)
The Hardware is a Raspberry Pi5 with 8GB RAM and SSD.

When i run PrusaSlicer from the shell i get some error-output:

/tmp/.mount_PrusaSXyInlr/usr/lib/aarch64-linux-gnu/libgnutls.so.30: version `GNUTLS_3_7_4' not found (required by /usr/lib/aarch64-linux-gnu/gio/modules/libgiognutls.so)
Failed to load module: /usr/lib/aarch64-linux-gnu/gio/modules/libgiognutls.so
[2024-10-24 16:41:17.901648] [0x00007ffef902a280] [error]   WebViewDialog error: wxWEBVIEW_NAV_ERR_OTHE

The installed gnutls version is: 3.7.9-2+deb12u3

Has this version of PrusaSlicer some hard dependecy on an older version of gnutls and the newer version is not backwards compatible?

davidk, 2024-10-27:
Hi there, please try the `-prusaconnect` AppImage that was just built recently.

This one should be most appropriate for the OS currently being run: https://github.com/davidk/PrusaSlicer-ARM.AppImage/releases/download/version_2.8.1/PrusaSlicer-version_2.8.1-aarch64-latestOS-prusaconnect.AppImage

The change for these AppImages was to unbundle libgnutls, as preferring the system libgnutls is more appropriate in this case

fberst, 2024-10-29:
I tried the new prusaconnect AppImage, and it works without issues.

Thanks :-)

closed 2024-10-29 by fberst