davidk/PrusaSlicer-ARM.AppImage #37: Package 'libwxgtk3.0-gtk3-dev' not found so impossible all different  "PrusaSlicer-version_2.9.0-aarch64"  appimage

dwprod38, 2025-02-03:
**Description of the issue**
All PrusaSlicer 2.9.0 ImageApp impossible to be installed and 'libwxgtk3.0-gtk3-dev' not existing...

**Hardware environment and OS that is being used to run PrusaSlicer-ARM.AppImage**

 - Hardware Raspberry Pi 5 (8GB) 
 - OS Linux raspberrypi 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

**Screenshots**

[image]

List of AppImage failed to be installed:

PrusaSlicer-version_2.9.0-aarch64-latestOS.AppImage
PrusaSlicer-version_2.9.0-aarch64.AppImage
PrusaSlicer-version_2.9.0-minimal-aarch64-latestOS.AppImage
PrusaSlicer-version_2.9.0-minimal-aarch64.AppImage

Cheers

davidk, 2025-02-05:
Hello,
For the OS version given, please install this package:`sudo apt-get install libwebkit2gtk-4.1-0`. The packages `PrusaSlicer-version_2.9.0-aarch64-latestOS.AppImage` or  `PrusaSlicer-version_2.9.0-minimal-aarch64-latestOS.AppImage` should be able to run once that is done

dwprod38, 2025-02-14:
Works fine thx !

davidk, 2025-03-11:
Closing this issue, thank you for confirming the solution

closed 2025-03-11 by davidk