davidk/PrusaSlicer-ARM.AppImage #38: Cross-compiling from x86_64 to aarch64

LaurensKrudde, 2025-03-10:
Hi, I was wondering if the container build can be used to cross-compile for aarch64 (pi5) using a x86_64 laptop. My laptop is running windows but I have WSL set up. I found some inspiration on the docker docs: https://docs.docker.com/build/building/multi-platform/#cross-compilation.

I've made some attempts at cross-compiling for aarch64, but without success so far. I am assuming you might have experience with this. Do you know how difficult this is? Is it as easy as setting a --platform parameter in the docker build or is more required to achieve this?

I know the github issues might not be the best place to ask this, but I couldn't find another way to contact you.

Thanks in advance.

davidk, 2025-03-11:
Hello! It is possible, but attempts in the past have mixed results as it is not perfect. The container build has usually run on native aarch64 processors instead.

For cross-compiling, the most condensed example might be in this GitHub action that was contributed some time ago: https://raw.githubusercontent.com/davidk/PrusaSlicer-ARM.AppImage/refs/heads/main/.github/workflows/artifact.yml - but this uses qemu which can be quite slow.

There are helpers to achieve native performance: https://github.com/tonistiigi/xx#dockerfile-cross-compilation-primer but this is a path that I have not gone down yet.

Good luck!

davidk, 2025-03-21:
Closing - hopefully the response has answered the question, please follow up if there's anything that can be clarified (regardless of the issue's status).

closed 2025-03-21 by davidk