davidk/PrusaSlicer-ARM.AppImage #40: Gtk-WARNING GLIBC version not found (latest release)

theofficialgman, 2025-06-01:
**Description of the issue**
Warnings of this type are printed in the log repeatedly
```
(prusa-slicer:3515): Gtk-WARNING **: 21:14:17.802: /tmp/.mount_PrusaSkkdLoL/shared/lib/libc.so.6: version `GLIBC_2.38' not found (required by /usr/lib/aarch64-linux-gnu/gtk-3.0/3.0.0/immodules/im-fcitx5.so)
```

**Hardware environment and OS that is being used to run PrusaSlicer-ARM.AppImage**

 - Hardware [e.g. Raspberry Pi 4 (4GB), Raspberry Pi 4 (8GB)]: Nintendo Switch
 - OS [e.g. Raspberry Pi OS (32-bit/armhf), Ubuntu Mate 21.04 (64-bit/aarch64)]: Ubuntu Noble 24.04

**Steps to reproduce the reported issue (if applicable)**
Run the appimage

**Screenshots (if applicable)**

theofficialgman, 2025-06-01:
CC @probonopd @Samueru-sama

probonopd, 2025-06-01:
One way to solve is to bundle _all_ needed libraries in the AppImage as described here:

* https://github.com/prusa3d/PrusaSlicer/issues/13653